MatthiasOs/VenusOs-ModbusTCP #10: Doku Lesen Modbus regsiter und werte bereich

OliOssi, 2018-04-29:
[Beschreibung Modus Werte Basis.pdf](https://github.com/CommentSectionScientist/wolfsbau/files/1958831/Beschreibung.Modus.Werte.Basis.pdf)
Hier meine Doku und beschreibung

CommentSectionScientist, 2018-04-29:
Wertebereich von 16 Bit unsigned integer ist von 0 bis 65.535. range von signed integer ist von −32.768 bis 32.767 
"resierung" -> Reservierung

CommentSectionScientist, 2018-05-01:
Im Bild unten stimmen die Grenzwerte meiner Meinung nach nicht.

OliOssi, 2018-05-02:
[Beschreibung Modus Werte Basis.pdf](https://github.com/CommentSectionScientist/wolfsbau/files/1966723/Beschreibung.Modus.Werte.Basis.pdf)

OliOssi, 2018-05-02:
Hallo 
gerade eine neue Version erstellt und eine Berechnung in Excel geschrieben:
[Beschreibung Modus Werte Basis V02.pdf](https://github.com/CommentSectionScientist/wolfsbau/files/1967641/Beschreibung.Modus.Werte.Basis.V02.pdf)
[project]/3_technical_offer/Modbus_PLC/CCGC-Modbus-TCP-register-list-2.12 OOS.xlsx

CommentSectionScientist, 2018-05-02:
kann geschlossen werden. ist sehr gut erledigt worden. Dokumente in die cloud hochgeladen

closed 2018-05-02 by CommentSectionScientist